tigard-tools/tigard #5: revise mode selection switch

securelyfitz, 2020-06-11:
Right now the mode selection switch is a DPDT but we're only using half of it:
SPI/JTAG: DO and DI separated
SWD/I2C: DO and DI connected with resistor

We have three features worth considering for the second half of the switch:
1. Adding an I2C pullup on SDA
Theoretically this is necessary for I2C to work, but due to FT2232H's poor i2c implementation it doesn't actually matter. There may be corner cases where it does, but for those cases, a user can put a resistor between pins 1 and 4 of the JTAG header to get this done.

2. Disconnecting MISO from the SPI/I2C header in I2C mode
Theoretically this causes issues for i2c chips that use pin2 as their address input. I haven't encountered issues, perhaps because most devices latch their address bits at powergood and they're ignored after. The fix to this is not connecting that pin to the target system, defeating the convenience of the I2C header. 

3. Swapping between TDO and TMS on pin 4 of the SWD header
This allows using the swd connector as a 10 pin arm jtag connector as well. 

Options 1 and 2 are choosing to protect against a unknown risks in the least frequently used interface that's avoidable with inconvenience. Option 3 is adding support for a frequently used feature for a large audience.

I'm leaning toward option 3, but only if both @esden and @fharding1 feel that it's worth the risk that avoiding options 1 and 2 would mitigate.

securelyfitz, 2020-06-11:
Wait, does this work using only one switch:
SPI/JTAG: DO connected to MOSI
SWD/I2C: DO connected to DI via swd resistor

Who knew switches were so hard. Maybe i'm just getting old.

securelyfitz, 2020-06-11:
More details:
MISO really does need to be disconnected, but luckily that can be done with only one switch like mentioned above:
> SPI/JTAG: DO connected to MOSI
> SWD/I2C: DO connected to DI via swd resistor

My current thought is that option 2 is mandatory but doable on the first switch, and option 3 is the best choice for the second leg of the switch.

securelyfitz, 2020-06-12:
updating - tms/swdio is pin 2 of the swd connector.

fharding1, 2020-06-14:
> My current thought is that option 2 is mandatory but doable on the first switch, and option 3 is the best choice for the second leg of the switch.

I think I agree, just to clarify:

The first pole will connect DO to either MOSI or DI via the SWD resistor, and the second pole will connect pin 2 of the SWD header to either TDO or TMS.

securelyfitz, 2020-06-14:
Just to remove ambiguity - When switched, both the SPI header and the JTAG header (and i guess swd too) should have the correct wiring. The JTAG header is primarily for jtag, but in general, the wiring harness will also be used when trying to flywire connect to spi tespoints - not the 8-pin header which will most likely be exclusively for clips and sockets.

So:
BD2 enters the ftdi via the level shifter, via what is now DI TDO MISO SWDIO
After change, it should be just DI

First Pole:
When in JTAG+SPI mode, the switch shorts DI to TDO MISO
When in SWD+I2C mode, the switch shorts DI to SWDIO, which connects to DO TDI MOSI via resistor, currently R16
**this affects the SPI, JTAG, and SWD headers**

Second Pole:
When in JTAG+SPI mode, Pin2 of the swd header is shorted to TMS
When in SWD+I2C mode, Pin2 of the swd header is shorted to SWDIO
**this only affects the SWD header. SPI and JTAG are unaffected**
 
Dilemma:
Ideally in I2C mode, there's no R16 resistor, but we need it for SWD.
Since we need the resistor, ideally, DI TDO MISO SWDIO would be SDA.
However, on UNIT and V0.0, DO TDI MOSI connects to spi header pin 8 and this works fine as SDA

I don't think this is an issue. I also don't think that truly fixing it has a solution that doesn't require more parts. Really, we could to a 4-way JTAG-SPI-I2C-SWD switch even though it be mostly redundant, with the advantage of reducing # of unique parts at the cost of space.

fharding1, 2020-06-15:
> Dilemma:
Ideally in I2C mode, there's no R16 resistor, but we need it for SWD.
Since we need the resistor, ideally, DI TDO MISO SWDIO would be SDA.
However, on UNIT and V0.0, DO TDI MOSI connects to spi header pin 8 and this works fine as SDA
> 
> I don't think this is an issue. I also don't think that truly fixing it has a solution that doesn't require more parts. Really, we could to a 4-way JTAG-SPI-I2C-SWD switch even though it be mostly redundant, with the advantage of reducing # of unique parts at the cost of space.

Could we add a SPDT solder jumper to the back of the board that normally routes DI through R16 but also allows you to cut the jumper, solder jump the other throw, and connect it directly?

[image]
[image]

securelyfitz, 2020-06-15:
i like this option. It does complicate the schematic - worth documenting in the schematic. But allows those who never use SWD and heavily use I2C to get the best I2C circuit.

commit 8bd1b2cf (2020-06-17): New header wiring, logic analyzer, silk docs

* Revised silkscreen to include: mode switch meanings, vtarget selection
  info, vref/vtarget isolation, flashrom command, and pin assignments
(closes #10)
* Consolidated VCC/VTGT/VREF naming (closes #9)
* Revised mode selection switch to support connecting SWD hack resistor
  as well as support using the SWD connector as a full CORTEX connector
(SWD or JTAG). Added solder jumper that allows bypassing 330-ohm SWD
hack resistor for I2C (closes #5)
* Added bitmagic logic analyzer test header for output pins (closes #4)
* Changed JTAG header ordering  [... cut]
closed 2020-06-17 by fharding1