tigard-tools/tigard #9: Clarify VCC/VTGT/TARGET naming convention

securelyfitz, 2020-06-14:
Switch is labeled "target", headers are labeled "VCC", LED is labeled "REF", documentation and wiring harnesses i used "VTGT"

It should be consistent. Not sure what it should be. Arguments for each:

VCC: This is what all the pinouts will have it named normally. We label it MOSI not DI or BD2, so why would we make the voltage pin any different?

VTGT: reinforces that you're attaching to someone else's voltage (potentially)

Vx: Clarifies that it's not certain what voltage you're attaching

Switch label: "Target" probably makes sense here; If we go with VCC for the header pins, then the whole switch should probably be labeled the 'VCC selector' or 'Vcc switch' and either Target or TGT is fine.

Once deciding, be sure to update:
- Switch input label
- Switch device label
- Header labels
- LED label
- Jumper wire lables
- Documentation

securelyfitz, 2020-06-14:
aslo: VREF on the bottom vref/vtarget cuttable jumper.

While i think the easiest answer is to just label everything as it is in the schematic, i'm not sure that's the best answer for the novice end users who might be using the silkscreen text to make decisions.

fharding1, 2020-06-14:
I'm not a fan of VCC because it's unclear whether you're talking about Tigard's VCC (5v) or the target's VCC (5v, 3.3v, 1.8v, target). I haven't seen Vx before and I think that would just confuse me.

VTGT seems like the best option. It will look a little less elegant on the pinouts but it should be ok.

fharding1, 2020-06-14:
Is there a label that would make more sense for the jumper on the bottom? VREF/VTGT communicates that while normally the target and reference voltage are the same thing, you can isolate them from each other by cutting the jumper.

commit 8bd1b2cf (2020-06-17): New header wiring, logic analyzer, silk docs

* Revised silkscreen to include: mode switch meanings, vtarget selection
  info, vref/vtarget isolation, flashrom command, and pin assignments
(closes #10)
* Consolidated VCC/VTGT/VREF naming (closes #9)
* Revised mode selection switch to support connecting SWD hack resistor
  as well as support using the SWD connector as a full CORTEX connector
(SWD or JTAG). Added solder jumper that allows bypassing 330-ohm SWD
hack resistor for I2C (closes #5)
* Added bitmagic logic analyzer test header for output pins (closes #4)
* Changed JTAG header ordering  [... cut]
closed 2020-06-17 by fharding1