tigard-tools/tigard #12: Change MOSI/MISO to COPI/CIPO to follow OSHWA guidance

securelyfitz, 2020-06-29:
https://www.oshwa.org/a-resolution-to-redefine-spi-signal-names

I have used Main and Subordinate whenever spelling out the acronyms, but happy to jump on this bandwagon since it's the first formal, collective statement on the matter. Will update my class materials to match.

Primary change would be silkscreen and documentation, but probably worth renaming nets to match too.

securelyfitz, 2020-06-29:
Hmm. On re-reading, it seems they'd recommend using SDI and SDO labels on those pins, but waffles on which is which. (are we naming it from the controller's perspective or the peripherals?)

The only way to eliminate that ambiguity is to use 4-letter acronyms, so i still think COPI/CIPO is best.

fharding1, 2020-06-30:
:+1: fixed net names, silk, and README. Agreed COPI/CIPO is best.

fharding1, 2020-06-30:
After we test the v0.1 boards we should increment the version number to v1. Then I will render and clean up the board image so it'll not have MISO/MOSI anymore.

commit 10a4fd81 (2020-06-30): Replace all instances of MISO/MOSI with CIPO/COPI

https://www.oshwa.org/a-resolution-to-redefine-spi-signal-names

Closes #12
commit 0e16d779 (2020-06-30): Replace all instances of MISO/MOSI with CIPO/COPI

https://www.oshwa.org/a-resolution-to-redefine-spi-signal-names

Closes #12
closed 2020-06-30 by fharding1