tigard-tools/tigard #15: explore feasibility of ice40 programming interface

securelyfitz, 2020-07-06:
ice40 needs ICE_CDONE and ICE_CRESET not currently connected:

from top to bottom, from FTDI perspective:
* ADBUS0 - ICE_SCK -> out
* ADBUS1 - FLASH_MOSI|IO0 -> out
* ADBUS2 - FLASH_MISO|IO1 <- in
* ADBUS3 - SPI_CS (optional, not used for programming) -> out
* ADBUS4 - ICE_SS_B -> out
* ADBUS5 - NC
* ADBUS6 - ICE_CDONE <- in
* ADBUS7 - ICE_CRESET -> out

Potential solution:
- BDBUS7 not currently attached to anything. no spare output buffer pins
- BDBUS5 is only used for JTAG/SWD SRST, which is only used by openocd, which is easily configurable via config file:
`ftdi_layout_signal nSRST -ndata 0x0020 -noe 0x0040` becomes `ftdi_layout_signal nSRST -ndata 0x0080 -noe 0x0040`
- BDBUS6 can connect to input buffer A6
- Input buffer input A6 can be tied to A1; the RX pin will be used as ICE_CDONE and see on both ADBUS1 and BDBUS6
- LA input can give access to all necessary pins

esden, 2020-08-08:
I have taken tigard V0.1 board and applied the modifications as proposed above.

Exactly as shown in this picture below:
[image]

I used a 2k2 resistor for the connection between BDBUS7 to BTBUS5.

I can confirm that this setup works and allows the use of iceprog to program an iCE40 Flash.
Example command invocation:
```
iceprog -I B bitstream.bin
```

A flywiring connection from Tigard to a ice40 target for the purpose of bitstream FLASH programming would be the following:
* GND - GND
* JTAG TDI - MOSI/COPI/IO0
* JTAG TDO - MISO/CIPO/IO1
* UART RX - CDONE
* JTAG SRST - CRESET
* JTAG TCK - CLK
* JTAG TRST - CS

Also tested if programming directly into the iCE40 SRAM works. I can confirm that it does. The following iceprog command can be used:
```
iceprog -I B -S bitstream.bin
```

Do not forget to disconnect the Flash CS jumper on your iCE40 target and to swap CIPO/COPI order, either with jumpers on your target or through wire swap. A wire swap would result in the following fly wire mapping from Tigard to iCE40 target:

* GND - GND
* JTAG TDI - MISO/CIPO/IO1
* JTAG TDO - MOSI/COPI/IO0
* UART RX - CDONE
* JTAG SRST - CRESET
* JTAG TCK - CLK
* JTAG TRST - CS

Regarding the pin mapping. It would be lovely to have a dedicated connector for this, but space is limited and ice40 is a fairly small target subset all things considered. I think the compromise where it is possible to make a logic analyzer connector to ice40 programmer cable is a reasonable trade off.

But in case a connector is considered in the future I would like to propose the following icebreaker-bitsy connector pinout:

[image]

This connector is a 0.1" (2.54mm) pitch 2x4 connector.

securelyfitz, 2020-08-08:
Seems like a small enough change that it's worth implementing the additional routing but not the header, though good to document it in case someone forks or revises. 
@fharding1 let me know if it's not as straightforward as it looks or it's not feasible for other reasons.

fharding1, 2020-08-14:
@esden were you able to take a look at 0e0eb68?

securelyfitz, 2020-09-04:
awesome. assigning this to me to add to the docs.

securelyfitz, 2020-09-05:
documented in b602d86

closed 2020-09-05 by securelyfitz