tigard-tools/tigard #16: find a cheaper alternative for the samtek swd and LA connectors

securelyfitz, 2020-07-28:
the samtek ones are $$ even in quantity over 1/4 of the whole BOM cost for the two least-used connectors.

full shroud ones would obscure pcb labels.
no shroud would end up with backwards cables. Not the end of the world, but not good for a device designed for beginners.

adafruit seems to be able to get them and resell them cheaper than samtek, maybe super high volume? https://www.adafruit.com/product/4048

fharding1, 2020-07-31:
Unshrouded $0.49/PCS at 100+ https://lcsc.com/product-detail/Pin-Header-Female-Header_Amphenol-ICC_20021121-00010C4LF_Amphenol-ICC-20021121-00010C4LF_C150517.html
Shrouded $0.65/PCS at 100+ https://lcsc.com/product-detail/Pin-Header-Female-Header_Amphenol-ICC_20021221-00010C4LF_Amphenol-ICC-20021221-00010C4LF_C180221.html
Unshrouded THT (would require PCB changes) $0.46/PCS at 100+ https://lcsc.com/product-detail/Pin-Header-Female-Header_Amphenol-ICC_20021111-00010T4LF_Amphenol-ICC-20021111-00010T4LF_C239059.html

Only thing I'm not sure about is whether the center notch on that shrouded connector is large enough.

Droid-MAX, 2020-08-01:
Maybe someone needs this.

https://github.com/Droid-MAX/tigard/releases/tag/v0.1-20200731

securelyfitz, 2020-08-06:
I have one of these: https://www.digikey.com/product-detail/en/cnc-tech/3221-10-0300-00/1175-1735-ND/4878946
$0.40 at 1000+. 

Physically about .5" wide - fits, but obscures JTAG and LA silk and there's almost no clearance by the corner mounting hole. Seems like moving 

14-pin https://www.digikey.com/product-detail/en/cnc-tech/3221-14-0300-00/1175-1738-ND/4878947 for $0.57
Don't have one of these in hand, but looks like it would overlap a couple resistors btwn LA and SPI connectors.

Combined, saves several dollars - almost 25% on BOM cost and likely worth a little effort...

fharding1, 2020-08-07:
How important is keying?

securelyfitz, 2020-08-07:
The purpose of shroud is either keying or protecting pins. since the board already has a ton of vertical pins, i think the protection part is minimal - so we're just doing it for keying.

The question is how important keying is. It's a great convenience, but LA and SWD are the two connectors *I* use least.

- Neither connector has a dangerous VCC/GND disaster waiting to happen by rotating wrong - while the SPI connector does, and we don't shroud that
- Samtec options add $$ for the least used connectors - 

I think if the choice is $amtec connectors or un-shrouded, we should go with un-shrouded. If there's a cheaper shrouded option with minimal redesign, i'd consider it.

Perhaps we should plan on unshrouded, then shrouded could be a crowdsupply stretch goal?

fharding1, 2020-08-07:
That sounds good to me. I can play around with switching to the shrouded header when I make the other changes. Is it normal to have tiers like that on crowdsupply?

securelyfitz, 2020-08-07:
it'd normally be a stretch goal, like "if we hit 500 preorders or $x, then we'll upgrade all boards to have shrouded swd connectors"

All that said, i'm still dwelling on:
> Neither connector has a dangerous VCC/GND disaster waiting to happen by rotating wrong - while the SPI connector does, and we don't shroud that

which is making me thing we should stop worrying, use unshrouded, and document part numbers for those who want to mod their own.

fharding1, 2020-08-07:
Fine with me!

securelyfitz, 2020-08-07:
Update: Found 10-pin connectors for $0.175 and 14-pin for $0.235 in full reels on alibaba. Working on securing them. 

https://www.alibaba.com/product-detail/1-27mm-pitch-SMT-shrouded-header_60682653524.html

Let's keep design as-is, use these, and recommend unshrouded for people looking to build their own but reduce cost. If the alibaba connectors don't end up happening, then we can choose samtec or unshrouded in the future as costs allow. Closing issue.

closed 2020-08-07 by securelyfitz