tigard-tools/tigard #20: VTGT indicator remains lit without Vref applied when data out pins are hot

securelyfitz, 2020-08-04:
Initial observation::
Connect tigard, LA cable, and bitmagic with Vtgt off: VTGT indicator off. Vref = 0.3v
Slide switch to 1.8v: VTGT indicator on. Vref=1.8V
Slide switch back to VTGT: VTGT Indicator on (but should be off). Vref=1.0V
Disconnect bitmagic (from usb, or disconnect LA cable): VTGT indicator off. Vref=0.0V

Theory:
current is leaking through the level shifters back to Vref. 

Experiment:
- attach steady 3.3v to a pin and measure Vref.
- measure Vref: 0.3V
- slide switch to 1.8v. Vref=3.3v
- slide switch back to VTGT: Vref=3.3v

all output buffer pins affected. No input buffer pins affected.

Implication:
if vref=vtgt, voltage on an output pin could leak through to vref, powering all tigard output pins, interfering with the target.

Resoluttion:
SN74LVC8T245 datasheet section 13.2:
"Keep /OE high until VCCA and VCCB are powered up"

VCCA is taken care of by FT2232H's /ENABLE
VCCB = Vref. 

Right now:
/OE = /ENABLE
Fix:
/OE = /ENABLE or !Vref

I haven't come up with a magic fix without an extra component:
can't use the level shifters cause we're controlling them
can't use the spare sw1 poles cause that won't work w/ vtarget

securelyfitz, 2020-08-05:
Workaround for test purposes: powering tigard from a USB power supply without data lines will power on tigard and level shifters, but not trigger /enable, keeping the level shifters in high-z mode.

fharding1, 2020-08-06:
Can we do something like this (with an extra MOSFET)?

/ENABLE is HIGH, VREF is LOW (FTDI has not booted, no VREF): /OE is HIGH, therefore level shifter is OFF.
/ENABLE is LOW, VREF is LOW (FTDI has booted, no VREF): /OE is HIGH, therefore level shifter is OFF.
/ENABLE is LOW, VREF is HIGH (FTDI has booted, we have a VREF): /OE is LOW therefore level shifter is ON.
/ENABLE is HIGH, VREF is HIGH (FTDI has not booted, we have a VREF): /OE is HIGH therefore level shifter is OFF.

https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgpABZsAoAMxGzRBTz3YUPc5ZThoSGJBT1shLih7sMgmbxQoaUKPQAe7FeFLakaQap0A-APIBRTe1a7ymY3FrtVAekv0ATtyXzdCvyoaDHh6ACV-OUEwYgUaKgTaKmxobDUE4XCWBEFcVXjkyFVE+JZU9KhMrWwwJVlsDGkEYmcdADUwiwAxazBsJGwaJQaWGhbjNwsAOQBBACEAGSsAZR8WIrXsPGK1RgBDABsAZwBTNXEIlWTtlyCnEuTyxJgEegBzKJYbzCN7+lXWFQOFx+gp+FQIPtjmcge9sgFfkDAvQgA

[image]

securelyfitz, 2020-08-07:
Logic checks out. My analog skills are too rusty to even think of second guessing the simulation.

fharding1, 2020-08-07:
I'm a little concerned that you measured 300mV on VREF, considering the BSH103 has a very low Vgs(th) of 400mV, but considering there don't seem to be any more issues with the VTGT LED lighting up when you touch a VTGT pin anymore, I think we should be safe after adding one or two more VREF pull-downs to keep it safely below Vgs(th).

I have all the components to prototype this on a Tigard, so I can give it a try too.

securelyfitz, 2020-08-07:
tested again and measured 130mv, not 300mv. Don't know if that was a typo or other conditions changed. Same behavior whether i connect to computer USB (with day, /enable asserted, VTGT LED off, level shifter undefined) as to a usb battery(no data, /enable deasserted, VTGT LED off, level shifter tristated)

Will be interesting to see if prototype results yeild something different.

fharding1, 2020-08-08:
Even with /OE shorted to 3.3v current can leak from an output pin such as TDO back to VREF. So it appears the level shifter does not go true high-z with it's output disabled, contrary to the datasheet. I guess this means I'm missing something.

[image]

fharding1, 2020-08-09:
Actual issue: because current can flow from the output to VREF when the level shifter output is enabled, the level shifter output enable signal latches high.

* With mode switch at 1v8, apply voltage to an output pin. **Since /OE is low and the level shifter is not high-z, current starts flowing from the output to VREF**
* Flip the mode switch to VTGT. The LED stays lit despite removing the 1v8 buck power, because current is still flowing from the output to VREF, keeping the level shifter high-z and "latching" VREF
* Manually forcing /OE high turns the LED off, and it stays off, even if the output pin remains hot.

Resolution:

Documentation. Maybe look into an ideal diode solution in the future or something like that. No easy solution.

closed 2020-08-09 by fharding1