tigard-tools/tigard #22: Clarify LA port's capabilites in documentation

securelyfitz, 2020-08-04:
Optimistically, you could connect a logic analyzer to the LA port, then use Tigard as just an mechanical adapter.

Realistically, any load on testpoints affects signal integrity. Connecting a clip to a wire to tigard, which connects to 4 other pins and a level shifter, which connects to a LA harness, which connects to a LA is really a big load.

I'm opening this issue to discuss how to document this expectation, as well as have a record for those looking for more details.

Possible wording:

> The LA port is best suited to monitoring push-pull signals where Tigard is an intended driver or receiver. It's tested and working well with UART, SPI and JTAG, and generally works with I2C. The expected use case is that you are using Tigard to communicate with a target, but need to debug that communication.
> 
> The LA port *may* be used as a passive mechanical adapter although this usage is not supported. For best results, power the Tigard board but disconnect VTGT from the target. Additional effort will likely be required to ensure signal integrity

fharding1, 2020-08-06:
Yeah to be honest I thought this might be a possible issue. I think how you worded it makes sense.

securelyfitz, 2020-08-06:
Sounds good. added to readme with 5075a9f2f9f69aa7f01e8a4ac6b9179c88f21eaa

closed 2020-08-06 by securelyfitz