tigard-tools/tigard #29: Adding ZIF Socket for TSOP-48 NAND Dumping

whid-injector, 2020-11-13:
Aloha there!
Considering that some months ago I have started working on something similar to Tigard (i.e. called Piadinaboard that is the natural evolution of two previous board I worked on #Focaccia-Board and #Burtleina-Board [1][2]) and as soon as my new baby daughter will allow me to complete it... I will release it in opensource anyway... I don't see the reason to keep this thought for myself only. 😃 Especially because I really like your FOSS approach ❤️ 

In my FT2232H based board, beside level shifters and various pinouts for the supported protocols (i.e. SPI, JTAG, UART, SWD)... I have also added a ZIF socket in order to attach TSOP-48 NANDs.

The FT2232H works perfectly for dumping NANDs by using tools like nand-dump-tool or yand.

I feel like Tigard without this feature is kinda missing a key factor* that will easily allow hw hackers to dump NANDS without messing with the classic FT2232H breakout boards + tons of Dupont wires.... or some expensive Programmer like Dataman/RT809H/etc.

*Don't read me wrong. I am not saying Tigard is not cool. The opposite. What I am suggesting is something like... Imagine a juicy triple cheesburger missing two slices of crunchy bacon. Well... the ZIF socket to allow NAND Dumping is the bacon. 🤣 

[1] https://github.com/whid-injector/Focaccia-Board
[2] https://github.com/whid-injector/Burtleina-Board

securelyfitz, 2020-11-13:
Aloha amico!

We thought a bit about how to add that capability, but decided it'd add too much cost and complexity for only a single use case.

Solid, reliable level shifting was a top priority for Tigard. We use all the level shifters in unidirectional mode, which works well for the protocols we end up using, but makes a mess when you start getting too fancy with custom bit banging options - we can't even swap port A and port B for UART/JTAG with our implementation.

In the end, we opted to go for better, easier support for the simpler interfaces versus the cost and size of adding more level shifters and switch logic that might get NAND support. I'll look forward to getting my hands on a Piadinaboard!

ciao!

closed 2020-11-13 by securelyfitz

whid-injector, 2020-11-14:
Yeah, I was just looking at your schematics and totally agree that with those unidirectional buffers doing bitbanging VS a NAND will be a PITA. 
Well... let's say that a juicy triple cheeseburger is already tasty enough 🍔 ❤️  😄

securelyfitz, 2020-11-16:
nothing wrong with ordering a side of bacon. :)