tigard-tools/tigard #33: AVRDUDE Configuration?

Matir, 2020-12-05:
To use Tigard with avrdude, a small avrdude.conf is necessary:

```
programmer
  parent "avrftdi"
  id = "tigard";
  desc = "Tigard interface board";
  usbdev = "B";
  sck = 0;
  mosi = 1;
  miso = 2;
  reset = 5;
;
```

(Strictly speaking, the sck, mosi, and miso lines are not necessary as they are the same as the parent device.)

This configures it to use port B on the FT2232, and uses sRST for the reset line.

I'm happy to send a PR if you think including this in the repo would be useful, and if you can tell me if it should be in the root (I don't think so) or what such a directory might be named.  I also have an OpenOCD config I can contribute after some more testing.

securelyfitz, 2020-12-07:
Awesome!
This is something i knew *should* be possible but just haven't had the time to figure out if the software existed.

I've been meaning to figure out a plan for organizing the documentation and keeping track of common config files. It probably makes sense to just have a directory full of all of them, but there's probably a good argument to having that in a separate repository too - i'm not sure.

I have several other doc edits in progress, will be pushing them soon. Can make this at the same time.

thanks!

Matir, 2020-12-07:
If it helps, this should be used with avrdude like this: `avrdude -C +tigard.conf` This causes avrdude to first read the system configs (which gives the definition of all the chips, as well as the parent "avrftdi" programmer), then read this one.

I've now programmed a dozen times or more using this config and it just works.  (AVR ISP is basically SPI, so no big surprise there.)

securelyfitz, 2021-03-24:
finally added this in e02208ca971daef9f25430be8d94606f7fdcc4f4. Thanks for making it work! Will add tigard.conf with #36

closed 2021-03-24 by securelyfitz