tigard-tools/tigard #35: Add Fiducials

G-Pereira, 2021-03-23:
Hello! Really nice project! I would like to suggest adding 3 non-collinear fiducials (3 corners of the board) so a pick and place like the Index (https://github.com/sphawes/index) can be used to assemble these boards.

fharding1, 2021-03-23:
When the boards were manufactured I believe we just put fiducial markings on the panel edge rails. Also, different manufacturers often have different requirements/expectations for fiducials. I'm not sure it makes sense to add them to master.

G-Pereira, 2021-03-23:
I've meant fiducials on the board itself so people can assemble at shops or at home. I would imagine you would like to do that as well so decrease manufacturing costs and increase sales and profit

fharding1, 2021-03-23:
If we are doing a large run, sending gerbers with fiducial markings would be unnecessary and potentially confuse the manufacturer. If the manufacturer is handling panelization, they will want to have control over the size/shape/position of the fiducials.

Fiducial markings are not necessary for hand assembly.

If someone does have a home pick and place machine, and happens to be assembling just one board instead of a panel, I would imagine they have the ability to add fiducial markings themselves (and to their liking).

G-Pereira, 2021-03-23:
Fair enough!

closed 2021-03-24 by fharding1

securelyfitz, 2021-03-24:
Just wanted to add - we optimized tigard for mass production and automated assembly, in contrast to [TIMEP](https://github.com/Matir/timep) which is very similar and optimized for hand assembly. Small batch automated assembly falls into a gap between the two, but I agree with @fharding1 that since fiducial requirements vary based on your assembly method, it's best to keep it out of the board design itself.

G-Pereira, 2021-03-24:
Me too, that's a very valid point!