tigard-tools/tigard #38: Hardware connector details in Readme.md

hattesen, 2021-03-28:
The readme.md is very well written and provides virtually all the information that is required to use the various interfaces, it is a sign of the high level of professionalism of this project, where many other similar projects fall short. 

One piece of information that I would like to see added is a reference to the physical connector matching each hardware interface (pitch, pins, rows and a connector part number), especially for the non-0.1" pin headers like the [I2C connector](https://github.com/tigard-tools/tigard#i2c), possibly including full specs (both connectors and wiring) for frequently used wiring harness. This will make it easier to locate suitable harnesses/cables or find the parts to make them up. 

On a side-note, the details of each interface should probably be moved out of readme.md and into separate interface.md pages, to make the readme.md more manageable, as it is likely to continue to grow over time.

securelyfitz, 2021-04-01:
Thanks! glad you found the readme useful.

I agree, mechanical specs would be really useful. I'll add them.

I started issue #37 to capture some of the pros and cons of keeping a flat readme vs breaking it out.

skochinsky, 2021-04-14:
+1
My cables arrived with some wires popping out of the connectors so it would be nice to know the "canonical" layout to put them back in proper order (yes I know I can track it back from the pinout but...)

securelyfitz, 2021-04-14:
@skochinsky The UART cable came with only VTGT, GND, TX and RX attached since these are what's used 99% of the time. Loose DTR, RTS, etc wires were included in case you wanted to use them, but not assembled to keep the uart wiring harness easier to use for the common case.

Right now the pinouts are listed [further below in the readme](https://github.com/tigard-tools/tigard#uart-1) and will likely be moved up into a single UART section in the refactoring for #37

skochinsky, 2021-04-15:
Thank for the explanation!