tigard-tools/tigard #39: Understanding LA

Kabron287, 2021-04-06:
According you explanations and schematic LA has 8 channels xPB0-xPB7.
Actually, only 4 channels really works with Sigrok Pulseview:
UART_RX-AD1
UART_CTS-AD3
UART_DSR-AD5
UART_DCD-AD6
In theory two more signals could be used:
BD2, BD6
And that is all, because rest of the signals are not inputs and could not be switched to inputs.
So, as long as PulseView is the only suitable SW, LA is 4 channels width.  

Or, I do not understand something?
[image]
[image]

securelyfitz, 2021-04-06:
The logic analyzer port isn't there to use tigard as a logic analyzer, it's there to use a logic analyzer to inspect the communication between Tigard and your target system.

See the [I2C example](https://learn.securinghardware.com/lessons/i2c/) - if you were working with an I2C device but were having trouble getting a response from the target device, or were getting unexpected/invalid/corrupt data, you could attach the logic analyzer and then examine the i2c protocol and perhaps identify timing or signal strength issues.

Let me know if there's anywhere in the documentation where I could to clarify this to avoid misrepresenting Tigard as a logic analyzer itself. Yes, there is an MPSSE logic analyzer driver in pulseview, but my experience with it has been frustrating at best.

Kabron287, 2021-04-06:
Thanks for answer.
>>Let me know if there's anywhere in the documentation where I could to clarify this to avoid misrepresenting >>Tigard as a logic analyzer itself. 
Obviously - nowhere, but my question is an example of this.

I guess, it could be stated out, say in form you answer me:
The logic analyzer port isn't there to use tigard as a logic analyzer, it's there to use **another** logic analyzer to inspect the communication between Tigard and your target system.

securelyfitz, 2021-04-06:
revised the readme text in e1927e9951568c4fb76dfcf3dc6b9c09ee046816
Do you think the wording is more clear now?

The MPSSE logic analyzer driver appears to work nicely, but my experience is that the timing and data rate has LOTS of jitter, making signals not come out anything like expected. Ideally there should be a better clock source on an FT2232H intended for logic analyzer usage - though that might be doable on the existing headers and may require some firmware...

Kabron287, 2021-04-06:
Crystal clear

securelyfitz, 2021-04-07:
great. You're not the first to ask a similar question, so glad to get it documented clearly.

closed 2021-04-07 by securelyfitz