tigard-tools/tigard #41: Unable to get UrJTAG to work with Tigard

skullhornet, 2021-04-09:
I have an existing setup where I am using the Olimex ARM-USB-TINY-H, it contains the same FTDI device as Tigard the: 2232H

From Urjtag I specify the following cable: 

jtag>cable FT2232 vid=0x15BA pid=0x002A driver=ftd2xx-mpsse interface=0

To use it with Tigard I set the voltage switch reference to 3V3, change the VID/PIDs and the interface number.

jtag>cable FT2232 vid=0x403 pid=0x6010 driver=ftd2xx-mpsse interface=1

However when I run the 'detect' command no chain is detected. Needless to say it works perfectly with the ARM-USB-TINY-H.

securelyfitz, 2021-04-12:
Thanks for reporting. Was able to (partially?) recreate. 

I didn't need to define driver=, but there seems to be a bug in urjtag - by default the frequency is set to 6MHz, but is going much faster than that - too fast to get a result for detect or discovery. 

Try this:

```
jtag> cable ft2232 vid=0x403 pid=0x6010 interface=1
Connected to libftdi driver.
jtag> frequency 100000
Setting TCK frequency to 100000 Hz
jtag> detect
IR length: 5
Chain length: 1
Device ID: 00010110001101010010001001001111 (0x1635224F)
  Unknown manufacturer! (00100100111) (/usr/share/urjtag/MANUFACTURERS)
jtag> 
```

While this works for me, It should set tck to 100khz, but my logic analyzer shows it is actually running tck at 500kHz.

I updated the readme with this info, Let me know if this doesn't work for you.

closed 2021-04-12 by securelyfitz

skullhornet, 2021-04-12:
Thank you for looking into this. Using your instructions I was able to get a proper detect.

skullhornet, 2021-04-13:
I just compiled the latest version of UrJTAG (2021.03) and things look like they are done properly now. I think it's the precompiled versions/upstream which have this error. I attached a capture from my Salea using the new software which produces a .sal file. There is only one channel: TCK. If you don't own a Salea you can install the software and take a look at the capture without the hardware device.

[tigard_tck.zip](https://github.com/tigard-tools/tigard/files/6305024/tigard_tck.zip)