tigard-tools/tigard #43: USB String

UweBonnes, 2021-04-30:
Hello,

does tigrad put some string in the usb descriptors so that it can be recognized?. That string is very helpfull with automatic device detection.

securelyfitz, 2021-04-30:
you should see manufacturer, produc, and serial in the usb descriptor:
```[523710.719667] usb 1-1.2.3: new high-speed USB device number 101 using xhci_hcd
[523710.830827] usb 1-1.2.3: New USB device found, idVendor=0403, idProduct=6010, bcdDevice= 7.00
[523710.830831] usb 1-1.2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[523710.830834] usb 1-1.2.3: Product: Tigard V1.0
[523710.830836] usb 1-1.2.3: Manufacturer: SecuringHardware.com
[523710.830837] usb 1-1.2.3: SerialNumber: TG1000160
[523710.855758] usbcore: registered new interface driver usbserial_generic
```
```
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0403 Future Technology Devices International, Ltd
  idProduct          0x6010 FT2232C/D/H Dual UART/FIFO IC
  bcdDevice            7.00
  iManufacturer           1 SecuringHardware.com
  iProduct                2 Tigard V1.0
  iSerial                 3 TG1000160
  bNumConfigurations      1
```

closed 2021-04-30 by securelyfitz

UweBonnes, 2021-04-30:
With the definitions you gave and the schematic, I tried to add the tigard definition to blackmagic. Please checkout my fixes branch and compile "blackmagic PROBE_HOST=hosted. Running with the mode-switch set to JTAG , "blackmagic -t" should scan for JTAG and print the found devices. In the SWD setting  "blackmagic -e" should do a jtag scan. '-e" stands for "external resistor". Feedback welcome!