tigard-tools/tigard #45: Pinout detection-discovering

skaox, 2021-05-17:
Hello,

is it possible to detect-discover pinout (UART, JTAG, SPI, I2C), fuzzing and command enumeration like [Busside](http://wiki.infosectcbr.com.au/index.php/Busside) ?

Thank you in advance

securelyfitz, 2021-05-17:
Detect/discover: no
spi fuzzing/command enumeration: yes, but not implemented
i2c address detection: yes, with PyFTDI
jtag command enumeration: yes, with URJTAG and other tools

Unfortunately, detect/discover is not really feasibe.
Detection algorithms depend on algorithms that try different combinations of inputs an outputs. In order to ensure Tigard's normal operation was exceptionally robust, we used unidrectional level shifters, meaning we cant change inputs to outputs and vice versa.
Busside doesn't have the same degree of level shifting and i/o protection as tigard, but is very flexible in how it uses pins. the JTAGulator does have significant I/O protection, but that comes at a much higher cost, and does lead to issues with some specific JTAG targets not working in 'jtag adapter' mode

There's no reason that command searching couldn't be implemented for SPI, I'm just not aware of an implementation. If i were to, i would implement it using PyFTDI.
PyFTDI does include a script to detect properly-wired I2C devices. URJTAG includes a 'discovery' command that will detect the length of all the data registers, helpful in determining their purpose.

closed 2021-05-17 by securelyfitz

skaox, 2021-05-17:
Noted ; a big THANK YOU for the answer and what you're doing.