tigard-tools/tigard #46: UART mode problem

sidprice, 2021-05-31:
Just fired up my Tigard for the first time (was a Crowd Supply backer) to use the UART feature and it does not appear to be working.

I am using Windows 10 and in Device Manager I see two COM ports. I have connected my terminal emulator to the first one (COM35) and it is set up for 115,200, 8N1 and no handshake.

Tigard is not connected to the target PCB, it is set to use internal power. When I press keys in the terminal I do not see any LED flashes and a scope on the TX pin shows it remains in a high state.

I think I have the set up right, but since this is a first use I could be mistaken.
Sid

Droid-MAX, 2021-05-31:
try another ports？

sidprice, 2021-05-31:
I tried another serial adapter and it works, Tigard does not.

fharding1, 2021-05-31:
The FT2232H exposes two serial ports to your computer. Only one is used on Tigard. You mention trying the first COM port, have you tried the other?

On my (Linux) computer, /dev/ttyUSB0 is the correct one to use, while /dev/ttyUSB1 does nothing.

sidprice, 2021-05-31:
I tried both COM ports, the TX LED never flashes.

fharding1, 2021-06-01:
Perhaps it's a [driver issue](https://learn.adafruit.com/adafruit-ft232h-breakout/serial-uart)? You might try on an OS X or Linux computer if you have one available. I believe every unit was loopback tested, so it would be very unlikely that you got a dud (but not impossible).

securelyfitz, 2021-06-01:
1. Have you installed ftdi drivers?
2. Are you sure you're talking to the right com port? (only the primary port can be used for UART)
3. What terminal emulator are you using?
4. What serial adapter *does* work?
5. Does it work for JTAG/SPI/ETC?
6. Does it work on a linux/osx system?

See the screenshot below, i chose the first com port of the composite device, and was able to connect with putty:
[image]

sidprice, 2021-06-01:
Thanks for the pointers - drivers are installed:

[image]

Yes I am sure I am using the right serial ports.

I am using SecureCRT from Van Dyke Software.

I have not tested other modes.

I do not have a Linux or OSX computer.

The issue is that the TX LED never flashes, regardless of connection to the target or power mode.

Terminal connection to my target works fine with AdaFruit CP2104 Friend, I have several and they all work fine.

securelyfitz, 2021-06-01:
Are all three LEDs - PWR, EN, and VTGT - all lit up?

If you 'view devices by connection' you can confirm that you're using the com port associated with port A.

CP2104 has a silicon labs chip which means it doesn't rule out driver issues.

can you try using putty to rule out software issues?

can you try using jtag or spi or i2c to rule out hardware issues?

sidprice, 2021-06-01:
>>> Are all three LEDs - PWR, EN, and VTGT - all lit up?
Yes

>>> If you 'view devices by connection' you can confirm that you're using the com port associated with port A.

Yes, confirmed.

>>> can you try using putty to rule out software issues?

Seems that is the problem. If I use putty the TX LED flashes.

I have no idea why SecureCRT doesn't work the same.

securelyfitz, 2021-06-01:
thanks for testing. It's possible there's multiple different layers of API access to the serial ports in windows, and whichever one SecureCRT is using is the issue?

Let me know if you have any additional issues.

closed 2021-06-01 by securelyfitz