tigard-tools/tigard #51: SWD config holds nRF in reset making debugging impossible

littlelailo, 2022-01-04:
Disclaimer:
I'm new to HW stuff so maybe this is sth completely obvious to anyone except for me, but it cost me two days to figure out so I thought I should share it here.

I tried to debug an nRF Dev Board over the P18 header using the tigard and I kept getting a "poll 4 timeout" Error during the init sequence in openocd. The reason why this happens is that the openocd config holds the nRF in reset and so it's not responding leading to OCD erroring out.

The solution is pretty simple:
Change `ftdi_layout_init 0x0018 0x05fb` to `ftdi_layout_init 0x0018 0x055b` in the init file to not pull up the reset pin during init. And after that everything works (also reseting the board via telnet works, so I don't think I broke anything, just changed the initial value of the pin)
(As a sidenote I think the config file is out of date, it should be `ftdi layout_init` instead of `ftdi_layout_init` - I got a couple of DEPRECATED warnings before changing that)

And I have to say: Thanks for providing such an awesome tool, it really helps me with my project!

securelyfitz, 2022-01-04:
Good catch. I had mostly been testing and using it with the JTAG harness and only connecting VTGT, GND, SWCLK, and SWDIO - leaving reset disconnected. Fixed in 6f935c497fd831a3b53ff4480f017a005695ca4d

What version of openocd are you using? The config files were originally made before v10. I've hesitated to fix the depricated parts without testing against a few older versions of openocd - i'd prefer to have one config file that works against V9-11 and throws deprecation warnings than a separate file for each version. I know i'll have to fix it sooner or later, so i'll keep this issue open.

Glad you're finding tigard useful!!

littlelailo, 2022-01-04:
I'm using v11 (commit from late Dec)

And yeah I also tried to only do it like this (VTGT, GND, SWCLK and SWDIO) using jumper but for some reason that's not working at all (only have success when connecting over the P18 debug header). If you have any ideas on how to debug this I would also immensly appreciate it cause the real target doesn't have a 10 pin header (it's an AirTag)

securelyfitz, 2022-01-04:
I haven't used more recent nordic dev boards, but iirc some of the older ones detected the presence of the debugger and disabled/disconnected the built-in debugger when attached. Could that be it?

littlelailo, 2022-01-05:
I mean it works over P18 and I don't see detecting logic on the [schematics](https://www.nordicsemi.com/-/media/Software-and-other-downloads/Dev-Kits/nRF52-DK/nRF52-Development-Kit---Hardware-files-2_1_1.zip) (PCA10040_Schematic_And_PCB.pdf)...

For me it seems like the nRF struggles to pull the SWDIO line low (first one is over P18 second one is using jumpers):
[image]
[image]
With that in mind I connected the jumpers and then tried to find a difference by measuring resistance between the two cortex headers (cause in theory all pins should be connected with 0 Ohm if my jumper connections are right). The only difference I found between the jumpers and the cable connection is that if I measure SWDIO on the P18 to the cortex header on the tigard I get 330 Ohm (R16 - tigard schematics) which is bypassed when I use the cable connection instead of the jumpers but I don't see how that could result in such a problem... (but then again I'm new to HW so maybe that's actually the issue?)
I would love to break out the cortex header directly and then connect the pins 1 by 1 but unfortunately I don't have a 1.25 mm to 2.54 mm board or cables that fit into the small pins

securelyfitz, 2022-01-12:
Closing this because the original issue, the SWD config, is resolved.
Opening #52 to address the SWDIO-on-2.54mm-header issue and #53 to address the I2C 'hack' implications