tigard-tools/tigard #52: SWDIO not directly accessible on JTAG header

securelyfitz, 2022-01-12:
The 8-pin 2.5mm JTAG header has the TDI pin labeled as SWDIO. 

Issue #51 identified that this isn't working as intended. While this does work in some cases, but is not ideal because it is tied directly to the always-on always-output level shifter, and only tied to the input through the 330 ohm resistor. The target has to drive SWDIO very strongly to have it's signal heard. 

Workaround 1: Always use the cortex header whenever possible for SWD
Workaround 2: choosing a lower voltage on the VTGT switch might make it easier for the target to 'overcome' the level shifter's output

The solution to this is hardware redesign, perhaps using a 3-pole switch to swap SWDIO and TDO on all the auxiliary headers. This might benefit I2C as well. Since SWD and I2C are lower-prioirity interfaces, this is unfortunately a low priority.

ikarus23, 2022-03-29:
Hi. It is probably best to update the documentation (the main `README.md`) until this is fixed. I've run into the same issue because the readme still sates "You can also use the TCK(for SWCLK) and TDI(for SWDIO) pins of the JTAG header" as if there were no difference.

PS: Thank you for this amazing and well documented tool!

securelyfitz, 2022-03-30:
Good catch, thanks @ikarus23. I removed one other reference but missed that one. Glad the documentation is helpful to you!

XMPPwocky, 2023-08-02:
For existing Tigards, there's a pretty straightforwards hack to at least get SWDIO exposed on an ordinary 0.1" header- use the (header pin unpopulated, at least on my Tigard) ICE_CDONE pin (labeled "??" in silk-screen) on the JTAG header. Cut the trace between it and the level shifter, and then reconnect it with a bodgewire to pin 2 on the Cortex debug header. Now you should be able to use the "??" pin on the JTAG header as SWDIO.

securelyfitz, 2023-08-02:
@XMPPwocky nice workaround! the pin isn't populated since the ice40 programming is such a niche feature, and because then the JTAG and UART headers are different pin counts.