tigard-tools/tigard #55: JTAG : TDO seems to be stuck at 1

pirrup, 2022-08-08:
When i do detect in urjtag i get error about TDO stuck at 1 or 0

Any idea what the issue could be ? i checked TDO and no short with VCC or Ground

securelyfitz, 2022-08-09:
I need some more details about your setup to understand what's going on.
If you've got an error saying TDO is stuck, its typically a wiring issue. Is the error stuck at 1 or stuck at 0? If you short TDO to ground, does it change to zero and when you short it to VCC does it change to 1? If so - then your configuration is right, you just need to get your wiring correct to. If not- then it's time to examine your configuration.

Either way - let me know more about your setup - what version of what tools, what commands you're running, and what your wiring setup to your target is. Then i might be able to spot a mistake, or if not recreate your setup to see what's going on.

pirrup, 2022-08-16:
jtag> cable ft2232 vid=0x403 pid=0x6010 interface=1
Connected to libftdi driver.
jtag> frequency 100000
Setting TCK frequency to 100000 Hz
jtag> detect

when tigard is set to VTGT (no power to target) and do detect response is: TDO seems to be stuck at 1
when tigard is set to VTGT (with power to target) and do detect response is: TDO seems to be stuck at 1

when tigard is set to 1.8 (so VCC comes from tigard)  and do detect response is: TDO seems to be stuck at 0

using latest urjtag on ubuntu

wiring : 
TCK -> TCK
TRST -> TRST
TDO -> TDO
GND -> GND
TMS -> TMS
TDI -.> TDI
VTGT -> VREF 1.8

securelyfitz, 2022-08-16:
Nothing jumps out at me to try, so a few more questions to understand the situation:
What's the status of the 3 leds in each of those scenarios? https://github.com/tigard-tools/tigard#debugging

What happens in scenario 1 or 2 when you take the tigard TDO wire and short it to ground?
What happens in scenario 3 when you short TDO to VTGT?
How much power is the target consuming on the 1.8v VTGT in scenario 3? Can you check if the voltage with a multimeter?

Is the target confirmed working with another JTAG adapter? 
Could you try using openocd with the same setup to see if you get the same behavior?

pirrup, 2022-08-17:
scenario 1 : no VTGT LED
scenario 2 : ALL LEDS WORKING
scenario 3 : ALL LEDS WORKING 

short to ground :
scenario 1 : stuck at 1
scenario 2 : goes to stuck at 0

short to VTGT:
scenario 3 : goes to stuck at 1

power in scenario 3 is 1.77V

target i cannot disclose publicly , but can be discussed offline .

securelyfitz, 2022-08-17:
So, looking at all the pieces, you've got:
1. the tool, urjtag
2. the configuration within the tool for your interface
3. the usb communication to the controller
4. the controller (tigard)
5. the level shifting on the controller
6. the wiring to the target
7. the target itself

your tests completely rule out 2, 3, and 5, and partially cover 4 and 6.

The best bet now is to try one or all of the folllowing:
- use openocd to see if that identifies any idcode, ruling out a software issue
- use a different jtag adapter if you've got one, to rule out a bad tigard
- use a different jtag target, to rule out the target as an issue

iBringsTheRuckus, 2025-03-29:
I'm having this same exact issue. I'm trying to connect to a Xilinx XC3S50A FPGA via JTAG.
```
UrJTAG 2021.03 #
Copyright (C) 2002, 2003 ETC s.r.o.
Copyright (C) 2007, 2008, 2009 Kolja Waschk and the respective authors

UrJTAG is free software, covered by the GNU General Public License, and you are
welcome to change it and/or distribute copies of it under certain conditions.
There is absolutely no warranty for UrJTAG.

warning: UrJTAG may damage your hardware!
Type "quit" to exit, "help" for help.

jtag> cable ft2232 vid=0x403 pid=0x6010 interface=1
Connected to libftdi driver.
jtag> frequency 100000
Setting TCK frequency to 100000 Hz
jtag> detect
warning: TDO seems to be stuck at 0
```
I have it connected to TDI, TDO, TMS, TCK, and GND.
I have a VTGT connected via VCCAUX which is a solid 3.26v right off the FPGA. I have checked TDO as well as all the other pins and they are also 3.26v. So what does TDO seems to be stuck at 0 mean??
I've spent hours trying to figure out what's going wrong and haven't been able to connect via JTAG with urJTAG or OpenOCD. I'm slowly losing my mind over this.

I have also tried disconnecting VTGT and setting the switch on Tigard to 3.3v with the same exact outcome. I don't have another JTAG board to try, but I'm about to buy one to see if it's an issue with my tigard.

I do want to mention my Jtagulator does pick up TDI, TDO, TMS and TCK correctly and does give a device ID. Although it's incorrect. Talking with Joe, he had this to say about the device ID. "Yes, this is a normal response. The Device ID might not conform to the JTAG specification (the vendor can set it however they want)"

securelyfitz, 2025-03-29:
TDO stuck at 0 means that the TDO pin only ever sees 0v. It could mean that urjtag/openocd is looking at the wrong pin, or that the device isn't outputting anything, or that the tigard isn't connected to the jtag input pins properly, or the tigard isn't speaking jtag properly.

What happens when you try shorting TDO to vtgt? That should tell us if the software stuff is all up and running and talking to tigard properly.

iBringsTheRuckus, 2025-03-29:
I verified TDO definitely has 3.26v at boot and after boot. All the the required jtag pins have holes in the PCB where I soldered in connectors and the soldering is solid and so is the connection via jumpers to Tigard with the supplied jtag harness.

Just so I'm clear, you want me to short TDO to VTGT on Tigard itself and then run detect?

securelyfitz, 2025-03-29:
Where are you measuring tdo? If tigard sees 3.26v on the tdo pin, and your vtarget is somewhere around that, you should be getting told tdi is stuck at 1

Is the "JTAG-SPI/I2C-SWD" switch on tigard set to the left selecting jtag mode?

What i would expect to see is disconnecting the tdo wire from the target and connecting TDO (as in - Tigard's TDO pin, not the target's tdo) to ground would give you "tdo stuck at 0" and to vtgt should give you "tdo stuck at 1" 

If it doesn't - you're either not talking to the right jtag interface, or your tigard is having electrical issues.

iBringsTheRuckus, 2025-03-29:
Tigard TDO to ground I get stuck at 0
Tigard TDO tied to VTGT I get stuck at 1

So before connecting Tigard I get 3.26v on TDI, TDO, TMS and TCK

When I connect Tigard and test voltages at the pins on the target device I lose the voltage on all TDI, TDO, TMS and TCK.

I'm honestly not sure what's going on at this point

[image]

[image]

[image]

[image]

securelyfitz, 2025-04-04:
do you have any other jtag target devices you could test your tigard on?