tigard-tools/tigard #56: Tigard stopped working with avrdude - all output pins stuck high?

jackw01, 2022-11-10:
I have successfully used Tigard with avrdude on Windows 11 to flash AVR microcontrollers in the past (last time about 5 months ago) and tried to do it again today, but it does not work. It appears that all the output pins are stuck high (at VTGT) and even the reset pin is never pulled down. I am wondering if this is a driver issue, the only thing that I've changed since it worked properly is installing Windows updates. Tigard currently shows up as two separate COM ports in device manager, both using the FTDI 2.12.36.4 driver.

```
C:\********>avrdude.exe -C +C:\********\tigard.conf -c tigard -vvv -p m88pb
executable_abspath = C:/********/avrdude.exe
executable_abspath_len = 39
executable_dirpath = C:/********
executable_dirpath_len = 27
sys_config = C:/********/avrdude.conf
sys_config_found = true

avrdude.exe: Version 6.99
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/*********/avrdude.conf"
             Additional configuration file is "C:\********\tigard.conf"

             Using Port                    : usb
             Using Programmer              : tigard
I avrftdi_open(714): Using device VID:PID 0403:6010 and SN '(null)' on interface B.
D write_flush(479): Setting pin direction (0x0000) and value (0x0000)
I set_frequency(210): Using frequency: 150000
I set_frequency(211): Clock divisor: 0x0027
D avrftdi_check_pins_mpsse(571): Using valid mask mpsse: 0x0000fff8
D avrftdi_check_pins_bb(535): Using valid mask bitbanging: 0x0000ffff
I avrftdi_pin_setup(645): Pin direction mask: 002b
I avrftdi_pin_setup(646): Pin value mask: 0000
             AVR Part                      : ATmega88PB
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom        65    20     4    0 no        512    4      0  3600  3600 0xff 0xff
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               flash         65     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               lfuse          0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               hfuse          0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               efuse          0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               lock           0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               calibration    0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               signature      0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : avrftdi
             Description     : Tigard interface board

D set_pin(243): Setting pin 3,5 (BDBUS3, BDBUS5) as RESET: low (high active)
D write_flush(479): Setting pin direction (0x002b) and value (0x0000)
D set_pin(243): Setting pin 0 (BDBUS0) as SCK: low (high active)
D write_flush(479): Setting pin direction (0x002b) and value (0x0000)
D set_pin(243): Setting pin 3,5 (BDBUS3, BDBUS5) as RESET: high (high active)
D write_flush(479): Setting pin direction (0x002b) and value (0x0028)
D set_pin(243): Setting pin 3,5 (BDBUS3, BDBUS5) as RESET: low (high active)
D write_flush(479): Setting pin direction (0x002b) and value (0x0000)
W avrftdi_program_enable(880): Program enable command not successful. Retrying.
D set_pin(243): Setting pin 3,5 (BDBUS3, BDBUS5) as RESET: high (high active)
D write_flush(479): Setting pin direction (0x002b) and value (0x0028)
D set_pin(243): Setting pin 3,5 (BDBUS3, BDBUS5) as RESET: low (high active)
D write_flush(479): Setting pin direction (0x002b) and value (0x0000)
W avrftdi_program_enable(880): Program enable command not successful. Retrying.
D set_pin(243): Setting pin 3,5 (BDBUS3, BDBUS5) as RESET: high (high active)
D write_flush(479): Setting pin direction (0x002b) and value (0x0028)
D set_pin(243): Setting pin 3,5 (BDBUS3, BDBUS5) as RESET: low (high active)
D write_flush(479): Setting pin direction (0x002b) and value (0x0000)
W avrftdi_program_enable(880): Program enable command not successful. Retrying.
D set_pin(243): Setting pin 3,5 (BDBUS3, BDBUS5) as RESET: high (high active)
D write_flush(479): Setting pin direction (0x002b) and value (0x0028)
D set_pin(243): Setting pin 3,5 (BDBUS3, BDBUS5) as RESET: low (high active)
D write_flush(479): Setting pin direction (0x002b) and value (0x0000)
W avrftdi_program_enable(880): Program enable command not successful. Retrying.
D set_pin(243): Setting pin 3,5 (BDBUS3, BDBUS5) as RESET: high (high active)
D write_flush(479): Setting pin direction (0x002b) and value (0x0028)
D set_pin(243): Setting pin 3,5 (BDBUS3, BDBUS5) as RESET: low (high active)
D write_flush(479): Setting pin direction (0x002b) and value (0x0000)
E avrftdi_program_enable(889): Device is not responding to program enable. Check connection.
avrdude.exe: initialization failed, rc=-1
             Double check connections and try again, or use -F to override
             this check.

D set_pin(243): Setting pin 3,5 (BDBUS3, BDBUS5) as RESET: high (high active)
D write_flush(479): Setting pin direction (0x002b) and value (0x0028)
I avrftdi_close(807): LED Mask=0x0000 value =0x0028 &=0x0000
D write_flush(479): Setting pin direction (0x0000) and value (0x0000)

avrdude.exe done.  Thank you.
```

securelyfitz, 2022-11-10:
I haven't had to debug avrdude before and I don't have it set up on a windows system, but your output looks nearly identical to when i run the same command with no avr device connected.

What's the status of the LEDs on your tigard? https://github.com/tigard-tools/tigard#debugging describes what they each mean and how to fix some common issues

Once they're all lit, my best guess is an issue with the level shifters. To test them, the easiest approach would be to jumper TX and RX on the UART port, and open a terminal to see 1) if you get back all the characters you send and 2) to make sure the TX and RX leds are flashing on tigard when you're doing it and 3) repeat for each voltage modes. If that works, then your tigard is probably working properly, because it means the FTDI, level shifters, and voltage regulators are all at least partly working.

Let me know how that goes. If your tigard works fine through all those tests but not avrdude, then its still possible your ftdi chip or level shifters are parlty damaged, but most likely you're dealing with a configuration issue/change.

jackw01, 2022-11-11:
All three LEDs are on. I jumpered the TX and RX pins and opened a terminal on port COM5 ("USB Serial Converter A") and everything works as it should in all voltage modes.

securelyfitz, 2022-11-11:
Thanks for checking. It seems everything is mostly functional. The UART loopback test is the easiest - but actually uses different pins than the AVR programming does. Those pins are shared with JTAG, SPI, I2C, and SWD. Have you used your tigard for any of those purposes? If so - testing any of them would confirm that the same pins are working with different software, narrowing down the issue to an AVRDUDE or target problem. 

Again - I'm not very familiar with debugging AVRDUDE, but my approach would be to hook up a logic analyzer and watch what happens when running it. If you've got a BitMagic, it's really well suited since you can hook it directly to tigard's LA port. If that's what you've already done to make your initial observation that the pins were all high - then I might recommend repeating it with the target connected and disconnected. Perhaps the target is what's broken and pulling I/O high?

jackw01, 2022-11-12:
I don't have a logic analyzer, but I was able to confirm that the SRST pin stays high when not connected to the target. I was able to program the target using an Arduino as ISP, so I don't think that's the issue. I tried changing the reset pin assignment in the tigard.conf file to each of the FT2232 pins connected to the output buffer and found that the only pins I could see a signal with are ADBUS0 (UART_TX) and ADBUS4 (UART_DTR), but not ADBUS2 (UART_RTS) or BDBUS0 through BDBUS5 for some reason. 

I tried manually pulling down ADBUS2 and BDBUS0-BDBUS5 with a 10K resistor while monitoring each of the corresponding output pins, and was able to confirm this way that the output buffer IC is not the problem and that the FT2232 output pins themselves are not functioning as expected.

securelyfitz, 2022-11-14:
Ok, that's really odd behaviour. Some - but not all output pins are not properly outputting. It could be the FTDI, the level shifter itself, the pullpup resistor network in between them, or the pcb traces. Though -  it could still also be a software/configuration issue since you're still using only avrdude to debug.

The best piece of info right now would be to see if JTAG, SPI, I2C, or SWD work with a different program, like OpenOCD, Flashrom, or something else. If they do - it's probably an avrdude issue. If they don't - it's probably a hardware issue.

If it is really a hardware issue, the next thing i'd do is look carefully at the board to see if any traces or pads are scratched, shorted, or otherwise damaged. I'd use a multimeter to make sure RN1 and RN2 are showing 100k ohms.

jordiorlando, 2022-11-19:
Hey Joe, it's been a while! I'm seeing the same behavior on Windows 11 and have verified it using my BitMagic. I ran the following command on both Linux and Windows: `$ avrdude -C +tigard.conf -c tigard -p m32u4`. The tigard.conf file is exactly the same as the one in the README in this repo.

Linux logic analyzer capture:
[image]

Windows 11 logic analyzer capture using @mariusgreuel's AVRDUDE fork for Windows:
[image]

Windows 11 logic analyzer capture using stock AVRDUDE for Windows:
[image]

securelyfitz, 2022-11-21:
@jordiorlando thanks for checking it out. I dug up a windows laptop and tested @mariusgreuel's 6.x and 7.0 on W10 and both worked fine (after i spent too long debugging my ICSP cable plugged in backwards) 
I don't have a w11 system to test on though - do either you or @jackw01 have a w10 system you could try? If not i can figure out how to get one up and running.

Thanks,
-joe

jackw01, 2022-11-22:
I don't have a Windows 10 system, but I can confirm that avrdude 6.99 worked on early 2022 builds of Windows 11.

jackw01, 2022-11-23:
I have the same issue with both 6.99 and 7 on my current Windows 11 build.

jordiorlando, 2022-11-24:
@securelyfitz I can try it on my work laptop which runs Windows 10, but I won't have access to that machine until a week from Friday. So I'll try it then, if it's still relevant.

mariusgreuel, 2022-11-24:
@jackw01 It appears that you are using a build of avrdude that contains my hacked-up version of libftdi that works with the stock FTDI drivers (as opposed to the ones requiring you to swap drivers to libusb). It seems that there is a bug that causes the chip to always operate on port A (matter of fact, the hacked library ignores the port). I will see that I get that fixed.

In the meantime, you can try forcing avrdude to use port B by specifying a serial number. If your serial number is `TG123`, you want to append a `B` for port B, i.e. `TG123B`. If you add that to your config via `usbsn = "TG123B";`, it might work.

Alternatively, you may use a MinGW build (which uses libftdi based on libusb) and then swap the FTDI driver for something libusb can work with, such as WinUSB.

securelyfitz, 2022-11-29:
@mariusgreuel Thanks! I got my hands on a W11 system, tried this, and it worked.

@jackw01 could you give this a try on your tigard? I'll bet this is the problem, not a hardware issue.

jackw01, 2022-12-16:
Yes, this fixed my issue.

closed 2022-12-16 by jackw01