tigard-tools/tigard #57: UART character translation issue

robotskirts, 2022-11-29:
I successfully used the Tigard in loopback, but I'm having issues connecting it to other USB to UART adapters.

I've got the Tigard, a silicon labs CP210x device, and a Prolific device. All three of them behave as expected in loopback. Typing A sends 0x41.

Connecting the SL and Prolific to each other has the same behavior. An A comes through as an A.

Connecting either of them to the Tigard has a different behavior though. Typing an A on Tigard and the other adapter will receive 0x5F. Typing an A on the other adapters and the Tigard receives 0x5F 0x00.

Any idea what's going on? 

I'm particularly confused by connecting the prolific to the silicon labs working exactly like they would if it was loop back. I'm not sure what the Tigard does differently.

robotskirts, 2022-11-29:
The issue here is I'm attempting to talk to an RS-232 device and not a TTL level device. Luckily everyone knows to test with a capital A which made it easy to find this answer: [How does 0x41 sent on serial Rs-232 be received as 0x5F?](https://qr.ae/prXZo9)

closed 2022-11-29 by robotskirts

securelyfitz, 2022-11-29:
Trust me, you're not the first to do exactly this! I last encountered it with a cheap 'rs-232' device that omitted the level shifter...