tigard-tools/tigard #58: Deprecated OpenOCD TCL Commands

a-gavin, 2022-12-19:
As of OpenOCD version v0.11.0+ several general adapter commands [have changed](https://github.com/openocd-org/openocd/commit/5280eb618a8cab4639f1eca567472db7e5024d13) and as of v0.12.0-rc1 FTDI commands [have changed](https://github.com/openocd-org/openocd/commit/1d4b252bb16ef823d8e98bd70fc323099033898b), deprecating the older versions. This affects commands listed in the README. The deprecated commands used in Tigard's Openocd cfg files do continue to work on newer versions, but they will be phased out at some point.

Currently, the general adapter commands are up to date in the SWD config but not the JTAG config. Neither uses the current FTDI commands. Will submit a PR which fixes this shortly.

ia, 2024-12-28:
@a-gavin, is this PR still actual? Maybe someone should be ping'ed for a review since they may just did miss the notification with your pull request...

Dear @securelyfitz, @fharding1, @mossmann, @esden, @ijliao. Sorry to bother you, but could you, please, check this PR? Thanks.

securelyfitz, 2024-12-28:
Thanks for the ping. I did miss the notification at first but have to admit that I have seen nad procrastinated on this one.

My biggest dilemma is that I've got students using both openocd 10 and 11, and now 12 too.  I need to get all my class materials in sync with the differing config files for each version, and then update the Tigard docs without wiping away the old versions.

On the bright side, a config file for Tigard does now ship with openocd along with all the other interface config files, which is what I should probably be pointing people to.

a-gavin, 2024-12-30:
> On the bright side, a config file for Tigard does now ship with openocd along with all the other interface config files, which is what I should probably be pointing people to.

Good to know. It looks like the config file in OpenOCD is JTAG-specific (or at least appears to be) and only available v12+, commit [here](https://github.com/openocd-org/openocd/commit/830d70bfc66ada2a68c73283b9e4fa4770d408ee).

That I recall, the deprecated version still works for v12. Maybe OpenOCD just needs an additonal SWD-specific Tigard interface config (in addition to renaming the existing config as the JTAG version).

securelyfitz, 2025-01-17:
I am updating my training materials to use openocd 12. Once that's done, i'll update the docs, and link back to the tags for the correct 10 and 11 configurations.
One remaining factor that i have meant to figure out - where should the swd-vs-jtag decision go? Should I go with separate tigard-jtag.cfg and tigard-swd.cfg files, or should it be generic and the board/device file contain the transport definition?

a-gavin, 2025-01-17:
That's great! I'm not an in-depth user of OpenOCD, so I can't really say. I've seen a couple of examples of the transport selected in the interface TCL file (e.g. [SWD](https://github.com/openocd-org/openocd/blob/master/tcl/interface/ftdi/ashling-opella-ld-swd.cfg)/ [JTAG](https://github.com/openocd-org/openocd/blob/master/tcl/interface/ftdi/ashling-opella-ld-jtag.cfg) for the same debugger and [here](https://codingfield.com/blog/2022-08/debug-bl60x-with-openocd-and-gdb/#openocd) for just JTAG, all FTDI-based). I've done the same have done the debuggers I've used. However, I'm not this is the "correct" method, given that [other config files](https://github.com/a-gavin/openocd-confs/blob/2d918c761328c4ed0326bc595fba4592d00b2a41/esp32c6.cfg#L4) define it in the board file.

PR #59 (2025-01-17): Update OpenOCD confs to reflect current tcl commands
Fixes #58
closed 2025-01-17 by securelyfitz