tigard-tools/tigard #61: Connect LA port to other Logic Analysers

svanimisetti, 2023-01-25:
I recently purchased Tigard from CrowdSupply. Its a very versatile device - and I am really liking it so far! Apologies for posting this question here - as I could not find a way to contact from SecuringHardware.com.

On your [I2C vidoe](https://learn.securinghardware.com/courses/applied-physical-attacks-with-tigard/lessons/i2c/), you demonstrated how to connect BitMagic probe to LA port on Tigard. Is there a way I could connect other logic analysers that I already own? I have a [Bitscope 10U](https://www.bitscope.com/product/BS10/) with a [SmartPort](https://www.bitscope.com/blog/DJ/08.png). Can I create an adapter between the D0-D8 pins on the Bitscope and connect it to the LA port on Tigard as shown in the [schematic](https://github.com/tigard-tools/tigard/releases/download/v1.1/tigard.pdf). Which protocols can I observe in the logic analyser when connecting to LA port?

Thanks for making this tool available as OSHW! Looking forward to your repsonse.

securelyfitz, 2023-01-26:
I'm glad you're finding tigard useful!

Yes, that's exactly what the LA port is for. While the bitmagic comes with the 14-pin harness for connecting the two, it should mate with any generic 14-pin, 1.27mm pitch, IDC-style connector. the 8 channels it connects to are the 8 channels used across all the different interfaces. some more details about the header are on the readme: https://github.com/tigard-tools/tigard#la

svanimisetti, 2023-01-26:
Looks like I missed the section on LA in the README! I think I might have to rig-up a 14-pin 1.27mm IDC to 26-pin 2.54mm IDC custom adapter. Nonetheless, I think this will be a great feature to help troubleshoot by peeking into all 5 protocols without disconnecting the target. Appreciate your prompt response. Will close the issue.

closed 2023-01-26 by svanimisetti

securelyfitz, 2023-01-26:
No problem! Do be careful - with the amount of wiring that might be involved, I've seen that attaching the logic analyzer harness is sometimes enough to make things noisy enough that they can't operate at full speed. Good luck.