tigard-tools/tigard #67: OpenOCD via SWD port STM32F407G-DISC1 board using Tigard

svanimisetti, 2024-02-09:
I am trying to use Tigard with OpenOCD to program/debug STM32F407G-DISC1 board using the onboard 6-pin SWD port. I have researched existing issues on this repo and [elsewhere ](https://www.allaboutcircuits.com/technical-articles/getting-started-with-openocd-using-ft2232h-adapter-for-swd-debugging/) on the internet and got past most initial issues. However, I am currently stuck trying to resolve an error. For completeness, I am documenting all steps I have taken so far to easily reproduce the error.

Remove CN3 jumpers from board (see page #15 of [STM32F407G-DISC1 user manual](https://www.st.com/resource/en/user_manual/dm00039084-discovery-kit-with-stm32f407vg-mcu-stmicroelectronics.pdf)) and connect Tigard 2x5 0.05" CORTEX header pins to STM32F407-DISC1 SWD port pins is as follows:
 - VTGT->pin#1
 - SWCLK->pin#2
 - GND->pin#3
 - SWDIO->pin#4
 - nRST->pin#5
 - SWO->pin#6

1. Installed WinUSB driver for interface 1 using [Zadig](https://github.com/pbatard/libwdi/releases/download/v1.5.0/zadig-2.8.exe) (#47). Updated from `FTDIBUS (v2.12.36.4)` to `WinUSB (v6.1.7600.16385)` [USB ID: `0403 6010 01`]
2. I am using [OpenOCD v0.12.0-2](https://github.com/xpack-dev-tools/openocd-xpack/releases/download/v0.12.0-2/xpack-openocd-0.12.0-2-win32-x64.zip). When I launch with default script that came with installation (`openocd.exe -f interface/ftdi/tigard.cfg`), I observed many errors.
> Error: session transport was not selected. Use 'transport select <transport>'
> Error: Transports available: jtag, swd
> Error: SWD mode is active but SWD_EN signal is not defined.
5. After reading the documentation and issues #54/#63, I realized that all three (interface, board and target) configurations needed to be customized for Tigard+OpenOCD+SWD for STM32 microcontrollers. I arrived at the following configuration file by combining the built-in script with notes from aforementioned issues.
```
adapter driver ftdi
adapter speed 1800
transport select swd
ftdi device_desc "Tigard V1.1"
ftdi vid_pid 0x0403 0x6010
ftdi channel 1
ftdi layout_init 0x0028 0x002b
ftdi layout_signal nTRST -data 0x0010
ftdi layout_signal nSRST -data 0x0020
ftdi layout_signal SWD_EN -data 0
#reset_config srst_only
#tigard does not support open-drain reset modes
#since its output buffer is always enabled
reset_config srst_push_pull trst_push_pull
source [find target/stm32f4x.cfg]
```
4. I ran the above script using `openocd.exe -f tigard_openocd_swd_stm32f4disc.cfg` - but got stuck at following error.
> Error: Error connecting DP: cannot read IDR
5. If I comment the last line for target definition, the OpenOCD session starts. However, I cannot run any standard commands from a telnet session. See following error.
> Open On-Chip Debugger
> > reset init
> invalid command name "reset"

If I use the on-board ST-LINK V2 interface using `openocd -f board/stm32f4discovery.cfg` (after reconnecting the CN3 jumper), I don't have any issues with flashing firmware or debugging. I would appreciate if you could provide guidance on how to resolve the above issue. Since this board does not have JTAG interface, I would like to find a way to use Tigard via OpenOCD with SWD.

securelyfitz, 2024-02-09:
There have been a few changes in the config files for openocd 11 and 12. I haven't tested them myself, but as of 11, tigard.cfg should be in interface/ftdi/tigard.cfg. You are correct, there's more than just the tigard cfg. stuff like swd vs jtag varies based on your target, so it can't be a part of the generic tigard cfg.

I'd recommend running something more like:
`-f interface/ftdi/tigard.cfg -f target/stm32f4x.cfg`
so you can stick to the generic files.

I have used tigard on an stm32 'blue pill' using openocd 10 and only vcc,gnd,swck, and swdio, so i know it should be possible.

my best guess at what's going on with your current issue is that tigard is properly configured, but for some reason, you're not getting the expected response from the stm32. That's why openocd gives you a prompt with none of the 'cpu' commands.

Can you try hooking tigard back up and using the 'targets' command in openocd?

Also - are the reset and swo pins necessary? If so - what are they connected to on tigard?

Do you know if your stm32 is actually running and not currently in reset for some reason?

svanimisetti, 2024-02-09:
I disconnected the nRST and SWO pins from Tigard - and tried using generic files using your suggestions. However, I get a different error.

```
>C:/apps/openocd-v0.12.0/bin/openocd.exe -f interface/ftdi/tigard.cfg -f target/stm32f4x.cfg
Open On-Chip Debugger 0.12.0 (2023-01-14-23:37)
...
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: stm32f4x.cpu: IR capture error; saw 0x0f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Invalid ACK (3) in DAP response
```

Perhaps generic/default configuration files are setup for JTAG interface. When I use my custom SWD configuration, OpenOCD runs, but the output of `targets` command is also empty. The STM32 is currently running as I flashed a simple blinky firmware to test/debug this issue. I am guessing that I need to manually add the target definition for stm32f4x to get this to work.

securelyfitz, 2024-02-09:
oh, oops. The generic file doesn't have the swd/jtag choice, and neither does the generic stm file.
I have had multiple inconclusive conversations about this - does the transport select go with the adapter or the target... I have generic, jtag, and swd config files for tigard sitting in my path and use all three of them on different occasions. the one that made it into openocd is the one without the transport selected.

svanimisetti, 2024-02-09:
I found that @ruswernet posed a [custom configuration for SWD](https://github.com/tigard-tools/tigard/issues/63#issuecomment-1819740933) on #63. He was using a rp2040. I tried using his configuration with the stm32f4x target. But even that gave the same original error I reported.

```
>openocd.exe -f tigard-swd.cfg -f target/stm32f4x.cfg
Open On-Chip Debugger 0.12.0 (2023-01-14-23:37)
...
Error: Error connecting DP: cannot read IDR
```

securelyfitz, 2024-02-09:
good news: you're not the only one with the problem
bad news: they didn't solve it

https://community.platformio.org/t/stm32-programming-with-ft2232h-via-swd-encounter-error-connecting-dp-cannot-read-idr/36766

svanimisetti, 2024-02-09:
Thanks! I was aware of this issue and researched the steps the OP documented in his post. Nonetheless, I have posted on that thread to see if the OP has an update!

svanimisetti, 2024-02-09:
Resolved the issue using notes from [here](https://electronics.stackexchange.com/questions/182763/how-to-use-an-external-st-link-v2-with-the-stm32f4-discovery-board/410840#410840) and [here](https://electronics.stackexchange.com/questions/230036/problems-connecting-st-link-v2-and-stm32f4-discovery-board). I updated the firmware on the discovery board using the [STM32CubeProgrammer](https://www.st.com/en/development-tools/stm32cubeprog.html) utility to V2J43M28. Also, made sure the board is only powered using VDD and the USB is disconnected from onboard ST-LINK V2.

Now I am able to use `tigard-swd.cfg` posted in #63 with standard target script `stm32f4x.cfg` (`openocd.exe -f tigard-swd.cfg -f target/stm32f4x.cfg`). Running `targets` via OpenOCD now lists the microcontroller.

```
Open On-Chip Debugger
> targets
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f4x.cpu       cortex_m   little stm32f4x.cpu       running
```

closed 2024-02-09 by svanimisetti