tigard-tools/tigard #69: SVG logo?

RagingRoosevelt, 2024-03-12:
Is there an SVG copy of the logo?  I was going to 3d print [this case](https://www.thingiverse.com/thing:5169952) for the Protocol Tool and the Bit Magic and wanted to embed the logo but I haven't seen SVGs available.

No worries if this was an active decision to not include.

securelyfitz, 2024-03-12:
here you go!

[image]

closed 2024-03-12 by securelyfitz

RagingRoosevelt, 2024-03-13:
Thanks!  [It worked really well](https://www.thingiverse.com/make:1163505) =D

[image]

securelyfitz, 2024-11-30:
Finally did mine! My first multicolor print. I think i should have done the logo smaller like yours.
[image]