tigard-tools/tigard #71: GD25Q64(B) reading issues

IvanGlinkin, 2024-06-09:
Hi team,
How are you? Hope you are doing super great.

I have a small issue/request regarding TIGARD reading flashroms.
I have a TP-Link RE-205 repeater (https://www.tp-link.com/ae/home-networking/range-extender/re205/) with a GD25Q64(B) on the board.
CH341 reads this chip without any issues, but TIGARD can't. May I ask you what can be a problem? Thank you
[image]
`flashrom -p ft2232_spi:type=2232H,port=B,divisor=4 -V                                                         1 ⨯
flashrom v1.3.0 on Darwin 23.5.0 (arm64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
flashrom was built with LLVM Clang 15.0.0 (clang-1500.0.40.1), little endian
Command line (3 args): flashrom -p ft2232_spi:type=2232H,port=B,divisor=4 -V
Initializing ft2232_spi programmer
Using device type FTDI FT2232H channel B.
Disable divide-by-5 front stage
Set clock divisor
MPSSE clock: 60.000000 MHz, divisor: 4, SPI clock: 15.000000 MHz
No loopback of TDI/DO TDO/DI
Set data bits
The following protocols are supported: SPI.
Probing for AMIC A25L010, 128 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L016, 2048 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L020, 256 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L032, 4096 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L040, 512 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L05PT, 64 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L05PU, 64 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L080, 1024 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L10PT, 128 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L10PU, 128 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L16PT, 2048 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L16PU, 2048 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L20PT, 256 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L20PU, 256 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L40PT, 512 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L40PU, 512 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L512, 64 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25L80P, 1024 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25LQ032/A25LQ32A, 4096 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25LQ16, 2048 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for AMIC A25LQ64, 8192 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25DF021, 256 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25DF021A, 256 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25DF041A, 512 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25DF081, 1024 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25DF081A, 1024 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25DF161, 2048 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25DF321, 4096 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25DF321A, 4096 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25DF641(A), 8192 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25DL081, 1024 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25DL161, 2048 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25DQ161, 2048 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25F1024(A), 128 kB: probe_spi_at25f: id1 0xff, id2 0xff
Probing for Atmel AT25F2048, 256 kB: probe_spi_at25f: id1 0xff, id2 0xff
Probing for Atmel AT25F4096, 512 kB: probe_spi_at25f: id1 0xff, id2 0xff
Probing for Atmel AT25F512, 64 kB: probe_spi_at25f: id1 0xff, id2 0xff
Probing for Atmel AT25F512A, 64 kB: probe_spi_at25f: id1 0xff, id2 0xff
Probing for Atmel AT25F512B, 64 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25FS010, 128 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25FS040, 512 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25SF041, 512 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25SF081, 1024 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25SF128A, 16384 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25SF161, 2048 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25SF321, 4096 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT25SL128A, 16384 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel AT26DF041, 512 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
[... 328 lines cut ...]
Probing for Winbond W25X64, 8192 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Winbond W25X80, 1024 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for XMC XM25QH64C, 8192 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for XMC XM25QU64C, 8192 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for XMC XM25QH128C, 16384 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for XMC XM25QU128C, 16384 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for XMC XM25QH256C, 32768 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for XMC XM25QU256C, 32768 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Zetta Device ZD25D20, 256 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Zetta Device ZD25D40, 512 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Unknown SFDP-capable chip, 0 kB: No SFDP signature found.
Probing for AMIC unknown AMIC SPI chip, 0 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Atmel unknown Atmel SPI chip, 0 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Eon unknown Eon SPI chip, 0 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Macronix unknown Macronix SPI chip, 0 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for PMC unknown PMC SPI chip, 0 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for SST unknown SST SPI chip, 0 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for ST unknown ST SPI chip, 0 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Sanyo unknown Sanyo SPI chip, 0 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Winbond unknown Winbond (ex Nexcom) SPI chip, 0 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Generic unknown SPI chip (RDID), 0 kB: RDID byte 0 parity violation. compare_id: id1 0xff, id2 0xffff
Probing for Generic unknown SPI chip (REMS), 0 kB: compare_id: id1 0xff, id2 0xff
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
Releasing I/Os`

securelyfitz, 2024-06-10:
I've used tigard on gigadevices flash chips before so it should work.

First, have you gone through all the steps listed at https://github.com/tigard-tools/tigard?tab=readme-ov-file#debugging

Also, be sure the jtag-spi/swd-i2c switch is to the left in jtag-spi mode.

The only thing i notice in your setup is that it says it's running the mpsse frequency at 60mhz. It's normally 30. Setting divisor=8 might help, or changing the frequency.

If that's all good, then some details about your wiring and setup - perhaps a photo - would help.

IvanGlinkin, 2024-06-10:
Hi @securelyfitz
Thanks for the rapid and extended reply.
Yeah, I read the debugging page and everything is working as intended. Please find attached several photos from my side as a PoC.
[image]
[image]
[image]

>Also, be sure the jtag-spi/swd-i2c switch is to the left in jtag-spi mode.
Definitely, it's switched on as JTAG-SPI
>The only thing i notice in your setup is that it says it's running the mpsse frequency at 60mhz. It's normally 30. Setting divisor=8 might help, or changing the frequency.
I tried divisor from 2 till 100 - not working unfortunatelly
>If that's all good, then some details about your wiring and setup - perhaps a photo - would help.
cable is common Apple wire, which came with the laptop. but it's not only for charging - as you can see mac can see the tigard. also i used it for alfa wi-fi adapter and it's working fine
For the clip - the common one (see photos) and it's fully working clip (see photos with ch341) 

Tigard for UART is working fine as i tested it with the same router. all is good, but for spi - idk

Please don't be hesitate to tell me what else i can share to troubleshoot this. thank you so much
Regards, Ivan

securelyfitz, 2024-06-11:
how are you powering the router? how are you keeping the cpu idle why you read the flash with the CH341? 
The biggest difference in the circuit is probably the level shifters and current limiting resistors - could it be that the CH341 is just overpowering the CPU SS,SCK, and COPI pins, while the tigard's protection features are (by design) not?

IvanGlinkin, 2024-06-11:
> could it be that the CH341 is just overpowering the CPU SS,SCK, and COPI pins, while the tigard's protection features are (by design) not?

I don't know man :) it's you baby - you should know better ;)

regarding the powering. I used different options - from 3.3, till 5 and 1,8 and even powering from the main power supply.

3.3 - you may see the blue LED is on (on the router)
[image]

5 - router is protected (LED is off)
[image]

1.8 - not enough power (LED is off)
[image]

VGTG, powering from original power supply (LED is on)
[image]

> how are you keeping the cpu idle why you read the flash with the CH341?

I believe it's doing automatically by ch341/flashrom. I just connect and it's working, just exactly in your ads here -  https://vimeo.com/478706987. Just connect and get the firmware.

p.s. let me try on another device to check if it's the specific of this chip, or TIGARD

securelyfitz, 2024-06-11:
in the video, i am holding the CPU in reset. How are you preventing the CPU from accessing the spi flash and ensuring there is no contention on the spi pins while you read it? The CH341 may only be working because it's overpowering (and in the process, frying) the cpu's spi pins. If this is the case- it's actually a feature of tigard, and a flaw of the CH341.
Also - i would strongly advise against having an open device connected to mains power like you've got in the fourth example.

IvanGlinkin, 2024-06-11:
sorry, doing nothing to prevent the CPU from accessing the spi flash. Will perform some research how to perform that. But if you could guide me, how it's possible with TIGARD, I would really appreciate it.
But the question is - why it's working with "W25Q128.V" chip (just plug in and read the flash), but not working with GD25Q64? 

Regarding CH341 - that could be the case, indeed. 

> Also - i would strongly advise against having an open device connected to mains power like you've got in the fourth example.

Thanks Joe.

securelyfitz, 2024-06-11:
It could be that on your second target, the SPI flash uses a separate power supply from the main CPU. If that's the case, when Tigard provides power, it powers up the winbond chip, but not the CPU, and is able to speak, unobstructed, to the flash chip. The second target could also have protection resistors on the SPI lines, so that tigard is able to easily overpower the CPU's traffic without risking damage to either.

Getting it to work on your repeater is all about target preparation and not much to do with tigard. My approach would be to find a datasheet for the SoC, figure out it's reset pin, and hold that low while reading the flash. Otherwise, pulling the chip off might be the easiest approach

securelyfitz, 2025-04-04:
closing this assuming it's resolved, either by holding the cpu in reset or something else. Reopen if thats not the case.

closed 2025-04-04 by securelyfitz