tigard-tools/tigard #76: Question, how does the FT232 exactly select modes? How does this board implement it in hardware

HasanTheSyrian, 2025-02-22:


securelyfitz, 2025-02-23:
It depends on what you mean by the modes.

The "mode switch" on the board that switches between jtag/spi and i2c/swd basically connects the poci/tdo pin to the pico/tdi pin with a 330 ohm resistor.
This is because both i2c and swd have a bidirectional data pin, so we need to be able to both read and write to it, but the FT2232H does not really support that.

If you mean switching between spi and jtag and i2c and swd for the different tools - that's all taken care of by the software tools. The FTx232H family all support MPSSE, which is their way of doing high throughput I/O. That's what PyFtdi, flashrom, openocd, and others all use to make the ft2232h speak their protocol.

Does this answer your question? Let me know if you want more details on any part.

-joe

HasanTheSyrian, 2025-02-23:
I should've been clearer on what I'm trying to do. Im trying to use an FT232 breakout board (CJMCU FT232H specifically) to try to program my Xilinx FPGA dev board. I'm trying to see if I can program/debug it + my Mac because I'll be studying a semester away from my Windows PC. 

I'm also trying to get familiar with the chip because I'll implement it as a USB-JTAG (where you can also switch the JTAG mode off and use the USB port as normal USB device) programmer on my custom FPGA development board.

My aim is for it to be recognized by Xilinx tools directly (w/o using OpenOCD etc). It seems that the switch on your board is a workaround for the FT2232 and the selection for JTAG is done via software (can/does the IC get programmed once to use JTAG?). Im also confused on this, it seems that pins of the same communication protocol are on different selections of the switch.

[image]

*my breakout board hasn't arrived yet so I haven't had the chance to try it but I'm trying to understand how everything works in general

securelyfitz, 2025-02-23:
Tigard uses the FT2232H, not the FT232H. They're similar, but since you're using a different board based on a different chip, this isn't the right place to find the answer.

You *should* bug xilinx to tell them to stop going out of their way to block certain hardware from working natively. You could look into an alternative like openfpgaloader which supports tigard, all of the ftx232h family, and xilinx fpgas. I haven't dealt with xilinx tools in ages, but something similar to [this](https://gist.github.com/rikka0w0/24b58b54473227502fa0334bbe75c3c1) worked for me with ISE 10+ years ago to get FT232H working natively.

More discussion and explanation about the mode switch can be found in issue #5, however that's not likely what you want since that's all about i2c and swd.

You're probably looking for how to use the ft232h to interface an fpga board, your best bet would be to study how it's done on a similar board. Icebreaker is one example, though it uses lattice fpga and open source toolchain and loader. Digilent makes xilinx boards that work natively with ISE but they used to be pretty cagey about how it was wired up.

closed 2025-02-23 by securelyfitz