tigard-tools/tigard #77: Flash error with ESP32 LyraT v4.3 Board

cody328, 2025-03-05:
I am trying to flash the ESP32 LyraT v4.3 board with Tigard, but I am unable to successfully flash the firmware.

The command idf.py -p COM7 monitor works properly, but the flashing command does not execute successfully.
I have attempted both methods: using version 3.3 of Tigard and providing an external voltage supply to the LyraT board. Additionally, I switched to the VTGT setting on the Tigard board.

Could anyone provide assistance or suggestions on how to resolve this issue?

securelyfitz, 2025-03-05:
i don't have any experience with that board or with idf.py, but some questions/ideas:

- are you getting any error messages?
- does it work with other programming devices on that board? if so, which ones?
- does it work with other boards using tigard? if so, which one?
- does it work with tigard and that board using a different flashing tool, like esptool or the web-based flasher?

that should help narrow down where the issue lies

cody328, 2025-03-18:
Sorry for late reply.
Can we discuss via whatsapp?

securelyfitz, 2025-04-04:
cody,
If you were able to resolve this could you share what the solution was in case others have a similar issue?