tigard-tools/tigard #80: How to Dump EEPROM Contents from Tigard for Flashing to Custom FT2232H Board's EEPROM

santhoshgandham, 2025-07-25:
Hi Team,

I'm currently working on a custom FT2232H-based debugger board that mimics the Tigard design. I'm using the same ICs (FT2232H) and have followed the same pin configurations and logic layout.

To ensure firmware/software compatibility, I would like to replicate the exact EEPROM contents of the Tigard board and flash them into my custom board.

Could you kindly share how I can:
1. Dump the complete EEPROM content from an original Tigard board, and
2. Use that dump to program another FT2232H-based device.

I’ve tried using `ftdi_eeprom` and `libftdi1`, but I'm unclear on how to extract the EEPROM as-is (including string descriptors, CBUS settings, etc.).

Any guidance, EEPROM binary/hex dump, or `.conf` file would be immensely helpful.

Thanks in advance!

Best regards,  
Santhosh

securelyfitz, 2025-07-25:
ftdi_eeprom is weird, it needs a config file to read or to write the
eeprom. Below is most of the config file I use. Be sure to change the
manufacturer and product strings, as well as the serial number. Details on
the serial number:
https://github.com/tigard-tools/tigard?tab=readme-ov-file#serial-numbers

-joe

> ftdi_eeprom --flash-eeprom tigard.conf

tigard.cfg:

vendor_id=0x403
product_id=0x6010

# Max. power consumption: value * 2 mA. Use 0 if self_powered = true.
max_power=500

###########
# Strings #
###########
manufacturer="CHANGEME"
product="CHANGEME"

###########
# Options #
###########
self_powered=false  # Turn this off for bus powered
remote_wakeup=false # Turn this on for remote wakeup feature
use_serial=true     # Use the serial number string

# Normally out don't have to change one of these flags
in_is_isochronous=false     # In Endpoint is Isochronous
out_is_isochronous=false    # Out Endpoint is Isochronous
suspend_pull_downs=false    # Enable suspend pull downs for lower power
change_usb_version=false    # Change USB Version
usb_version=0x0200          # Only used when change_usb_version is enabled

eeprom_type=0x56

########
# Misc #
########

# This is the relative filename that EEPROM contents will either be
# read from or written to, depending on whether ftdi_eeprom is run
# with the --read-eeprom or --flash-eeprom option.
filename="eeprom.bin"

serial="CHANGEME"

santhoshgandham, 2025-07-25:
Thanks a lot for the for the replyvand I've programmed the eeprom of my custom board Firmex with this as you said and named it with tigard.conf:  

vendor_id=0x403
product_id=0x6010
max_power=500
manufacturer="SecuringHardware"
product="Tigard V1.1"
self_powered=false
remote_wakeup=false
use_serial=true
in_is_isochronous=false
out_is_isochronous=false
suspend_pull_downs=false
change_usb_version=false
usb_version=0x0200
eeprom_type=0x56

serial="TG110001"

I'm trying to connect to a target microcontroller (LM3S series ARM Cortex-M3) via JTAG using OpenOCD. I have two FT2232H-based boards:
1. **A Tigard board**
2. A custom **Firmex** Debugger board (VID:PID 0403:6010, serial TG110001)

**My goal is to interrogate the JTAG chain and dump data.**  I’m using the following OpenOCD config file (jtag_dump.cfg):

adapter driver ftdi
transport select jtag
ftdi_vid_pid 0x0403 0x6010
ftdi_channel 1
ftdi_layout_init 0x0038 0x003b
ftdi_layout_signal nTRST -data 0x0010
ftdi_layout_signal nSRST -data 0x0020
reset_config trst_and_srst separate srst_nogate trst_push_pull srst_push_pull
adapter speed 500
source [find target/stellaris.cfg]

**✅ With Tigard, the following output is seen:**

**$ sudo openocd -f jtag_dump.cfg**
Open On-Chip Debugger 0.11.0
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 500 kHz
Info : JTAG tap: lm3s.cpu tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x4)
Info : lm3s.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for lm3s.cpu on 3333

**❌ With Firmex Debugger, using the same config and wiring:**

**$ sudo openocd -f jtag_dump.cfg**
Open On-Chip Debugger 0.11.0
Info : clock speed 500 kHz
Warn : There are no enabled taps
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: lm3s.cpu: IR capture error; saw 0x0f not 0x01
Error: Invalid ACK (7) in DAP response

lsusb -v -d 0403:6010 | grep -E 'iManufacturer|iProduct|iSerial'

  iManufacturer           1 SecuringHardware
  iProduct                2 Tigard V1.1
  iSerial                 3 TG110001

**What I want to understand:**
Why does the Tigard board work flawlessly with the same jtag_dump.cfg, connections, and target, while the Firmex debugger fails?